eclipse-leda/leda-distro #3: git submodules are not available

phongt, 2022-08-22:
There is  `meta-sdv` in [.gitmodule](https://github.com/eclipse-leda/leda-distro/blob/main/.gitmodules#L27) can not find.

`
$ git clone https://github.com/eclipse-leda/leda-distro.git
$ cd leda-distro
$ git submodule init
$ git submodule update
`
 After executing `git submodule update` can not find any git submodule (eg: poky, meta-virtualization...). Look like missing commit & push the git submodule directory

mikehaller, 2022-10-06:
Thanks for bringing this up.
We're currently migrating to kas, which does the submodule handling automatically.

phongt, 2022-10-15:
cool!.I can build with instruction in [README.md](https://github.com/eclipse-leda/leda-distro/blob/main/kas/README.md)

closed 2022-10-15 by phongt