eclipse-leda/leda-distro #13: New releases available

el-paso, 2022-11-29:
I tried to test leda according to https://eclipse-leda.github.io/leda/docs/general-usage/download-releases/

There is no realese available.

mikehaller, 2022-11-29:
We're working on setting up the build environment. As the requirements for a Yocto-based build exceeds the limitations of GitHub-provided runners, it takes much longer than initially anticipated.

Since you are a registered committer to the project, you may have some helpful ideas on how to provide the Eclipse Leda project with some larger GitHub-managed runners. We'd be very happy if there would be a good way.

mikehaller, 2022-12-06:
We still do not have an official release yet, but during testing of our build and release workflow, we made a test-build.
Available here: https://github.com/eclipse-leda/leda-distro/releases/tag/0.0.3

chheis, 2022-12-22:
Test Releases available @el-paso

closed 2022-12-22 by chheis