eclipse-leda/leda-distro #18: Home/End key navigation does not work with putty

int0x27, 2022-12-02:
Leda image uses `TERM=xterm`, which is causing issues in command line editting (home and end keys do not work).
`export TERM=linux` fixes that isses.
As `$HOME/.bashrc` is not used, probably put it in `/etc/profile`

mikehaller, 2023-01-16:
It seems to depend on your host terminal.

Whether you're using the serial console when starting qemu, then it's already TERM=linux

When connecting via ssh, it's TERM=xterm-256color on my terminal, since my host terminal is using xterm-256color.

If you're using a SSH .config, then just set `SetEnv TERM` on the respective `Host` section.

chheis, 2023-03-02:
@int0x27 still unclear? Or can I close the issue with solved?

int0x27, 2023-03-06:
It was not an actual issue, It probably affects putty only and it can be fixed in the putty session.
You may close it

closed 2023-04-14 by mikehaller