eclipse-leda/leda-distro #20: Integrate Eclipse Kuksa.VAL - GPS Feeder

mikehaller, 2022-12-06:


mikehaller, 2023-01-16:
This is has been blocked due to compilation errors. Needs investigation.

mikehaller, 2023-06-21:
For the default configuration, Leda should use a "sane default".

In case of [gpsd](https://wiki.ubuntuusers.de/gpsd/), it seems we should be using `/dev/ttyUSB0` as default, as it supports most of the GPS devices out of the box.

d-s-e, 2023-07-07:
According to [your comment here](https://github.com/eclipse-leda/leda-distro/issues/128#issuecomment-1625003195) `gpsd` should run inside the container and not on the host, correct?

mikehaller, 2023-07-08:
I would say yes, as it is an optional feature from an SDV point of view, and it's easier to configure at runtime via the container runtime.

Since our current supported HW platform is not static (eg raspi gps module can be attached and is not built in)

But i think it would be acceptable tradeoff to have gpsd preinstalled natively, once we know how to dynamically change the config at runtime through Desired State. Since that is conceptually not clear yet.

So with sane default for Raspi, we could just start by preinstalling it on the sdv-image packagegroup additions.

We could use this issue also for testing out the bitbake-built container images.

wdyt?

closed 2023-09-12 by vasilvas99