eclipse-leda/leda-distro #31: Documentation for desired state deployment

mikehaller, 2023-01-09:
The current documentation is outdated, as it is based on k3s deployment descriptors:
https://eclipse-leda.github.io/leda/docs/device-provisioning/vehicle-update-manager/

It needs to be updated to the new desired state message format and provide an example deployment document as well:

The example deployment document should be provided as an example/template within the leda-distro as well, so it's easy to copy and adapt.

```
mosquitto_pub -t vehicleupdate/desiredstate -f my_deployment.json
```

my_deployment.json:
```
{
	"activityId": "my-app-poc-activity",
	"payload": {
		"domains": [
			{
				"id": "containers",
				"components": [
					{
						"id": "vehicle-update-manager",
						"version": "v0.0.2",
...
```

mikehaller, 2023-01-16:
The implementation on the VUM is ongoing and there seems to be a major change in the spec. It would be good to wait until that implementation is available before documenting the old version.

vasilvas99, 2023-10-03:
This issue is now stale and superseded. There multiple method to deploy containers on leda now (see KAD, VUM)

closed 2023-10-03 by vasilvas99