eclipse-leda/leda-distro #51: Script-based Leda-Setup

JochenKienzle, 2023-02-23:
**Is your feature request related to a problem? Please describe.**
I want to setup the components provided by the leda-distro on an existing (and supported) operating system (e.g. ubuntu server) instead of replacing the complete OS which might have specific configurations and other components.

**Describe the solution you'd like**
A 'provisioning script' which installs the 'leda-stack' on a supported operating system. (Downloading needed base components, setup needed configurations, setup services).

**Describe alternatives you've considered**
- Using docker images of leda on an existing operating system but this comes with additional overhead and docker images for architectures like arm64 are currently not available

**Additional context**
Such a script would decouple the dependency to the very specific operating system and allows are more integrative approach for existing systems and could also speed up the 'initial setup'-process (no flashing needed --> Only execution of install script).

mikehaller, 2023-06-22:
There is now a very basic [install-leda.sh](https://github.com/eclipse-leda/leda-distro/blob/main/resources/debian/dockerfiles/install-leda.sh) script, which works on Debian 11+ and Ubuntu 20+. Probably runs on Raspi as well.

It only installs containerd, Kanto, Leda Utilities and the container deployment manifests for the core components and examples.

It's missing any additional setup (certificate generation, device provisioning, CAN-bus or network setup etc.), but at least it's a start.

Next step would be to set up a proper APT repository somewhere (github?) and push the .deb files there.

I would consider this issue closed as soon as we have something like `curl http://some-nice-leda-website.io | sh` working on most systems.

wdyt?

JochenKienzle, 2023-06-23:
@mikehaller Sounds great and many thanks for providing it 👍