eclipse-leda/leda-distro #56: `sdv-provision` produces an empty  `Device ID`

razr, 2023-03-03:
**Describe the bug**
`sdv-provision` produces an empty  `Device ID` and exits afterward.

```bash
root@qemux86-64:~# ping 8.8.8.8
64 bytes from 8.8.8.8: seq=1 ttl=115 time=2.368 ms
64 bytes from 8.8.8.8: seq=2 ttl=115 time=2.343 ms

root@qemux86-64:~# sdv-provision
Checking Eclipse Leda Device Provisioning configuration...
- Certificates directory exists
Checking Device ID
- Based on network device: enp0s2
- Device ID:
Checking whether either IdScope or ConnectionString is configured
- Id Scope file found: /data/var/certificates/azure.idscope
- Id Scope configured in cloudconnector deployment descriptor /data/var/containers/manifests/cloudconnector.json
Checking device certificates
- All device certificates are present
- Primary device certificate: /data/var/certificates/device.crt
- Primary device private key: /data/var/certificates/device.key
- Secondary device certificate: /data/var/certificates/device2.crt
- Secondary device private key: /data/var/certificates/device2.key
Fingerprints (add these to the Azure IoT Hub Device)
- Primary thumbprint: 7CBAA9FBD0AB6CDD82DBDAA482BC90827F4ED265
- Secondary thumbprint: 3F6FD1CFB8E7DB38CD768C19FA8D8F1E5590F198

root@qemux86-64:~# cat /etc/deviceid
```

**To Reproduce**
Steps to reproduce the behaviour:
1. `./run-leda.sh`
2. `sdv-provision`

**Expected behaviour**
A valid `Device ID` shall be generated?

**Screenshots / Logfiles**
N/A

**Leda Version (please complete the following information):**
 - Version: [2022, 0.0.4](https://github.com/eclipse-leda/leda-distro/releases/download/0.0.4/eclipse-leda-qemu-x86_64.tar.xz)
 - Machine: qemux86_64
 - Connectivity: corporate network

**Additional context**
@mikehaller I have followed this doc https://eclipse-leda.github.io/leda/docs/device-provisioning/script-provisioning/. Do I need to create an account here:  https://azure.microsoft.com/free/iot

mikehaller, 2023-03-16:
It seems that on the first run of the script, the network was not up and created an empty /etc/deviceid file.
On second run, the network was already up, but the deviceid file was already empty.

Bug needs to be fixed in the sdv-provision script to check if the deviceid file is empty and retry to recreate it with proper id.

On Raspi, we may think about a fallback to use the HW serial number in case there is no MAC address yet?

razr, 2023-03-16:
if I remove `/etc/deviceid` before I run `sdv-provision` for the first time I have another error and the `deviceid` is empty anyway.

```bash
sed: /address: No such file or directory
```

```bash
root@qemux86-64:~# sdv-provision
Checking Eclipse Leda Device Provisioning configuration...
- Certificates directory exists
Checking Device ID
- Based on network device: enp0s2
- File does not exist, creating: /etc/deviceid
sed: /address: No such file or directory
- Device ID:
Checking whether either IdScope or ConnectionString is configured
- Id Scope file found: /data/var/certificates/azure.idscope
- Id Scope configured in cloudconnector deployment descriptor /data/var/containers/manifests/cloudconnector.json
Checking device certificates
- All device certificates are present
- Primary device certificate: /data/var/certificates/device.crt
- Primary device private key: /data/var/certificates/device.key
- Secondary device certificate: /data/var/certificates/device2.crt
- Secondary device private key: /data/var/certificates/device2.key
Fingerprints (add these to the Azure IoT Hub Device)
- Primary thumbprint: 7CBAA9FBD0AB6CDD82DBDAA482BC90827F4ED265
- Secondary thumbprint: 3F6FD1CFB8E7DB38CD768C19FA8D8F1E5590F198
```

it looks like `deviceid` should be in my case
```bash
sed 's/\:/\-/g' /sys/class/net/enp0s2/address
52-54-00-12-34-02
```

so I have created it manually
```bash
sed 's/\:/\-/g' /sys/class/net/enp0s2/address > /etc/deviceid

root@qemux86-64:~# sdv-provision
Checking Eclipse Leda Device Provisioning configuration...
- Certificates directory exists
Checking Device ID
- Based on network device: enp0s2
- Device ID: 52-54-00-12-34-02
Checking whether either IdScope or ConnectionString is configured
- Id Scope file found: /data/var/certificates/azure.idscope
- Id Scope configured in cloudconnector deployment descriptor /data/var/containers/manifests/cloudconnector.json
Checking device certificates
- All device certificates are present
- Primary device certificate: /data/var/certificates/device.crt
- Primary device private key: /data/var/certificates/device.key
- Secondary device certificate: /data/var/certificates/device2.crt
- Secondary device private key: /data/var/certificates/device2.key
Fingerprints (add these to the Azure IoT Hub Device)
- Primary thumbprint: 7CBAA9FBD0AB6CDD82DBDAA482BC90827F4ED265
- Secondary thumbprint: 3F6FD1CFB8E7DB38CD768C19FA8D8F1E5590F198
```

Thanks for helping with it!

closed 2023-06-29 by d-s-e