eclipse-leda/leda-distro #60: Problem with manifest files in 0.0.5 Pi Image

walzerT-onsite, 2023-03-14:
**Describe the bug**
I recognized that databrooker and feeder are not visible in kantui. So I tried to deploy them. And that gives me the information that "wrong json in *" all files that are in manifests. A bit strange but alle files that are in manifests_dev are deployed right. 

The first difference that I noticed was:

feedercan
```
    "container_id": "feedercan",
    "container_name": "feedercan",

```
seatservice (that working):
```
    "id": "seatservice-example",
    "name": "seatservice-example",
```

**To Reproduce**
Steps to reproduce the behaviour:
1. Check if the containers in kantui visible 
2. kanto-auto-deployer /data/var/containers/manifests

**Expected behaviour**
I would expect that the images  in manifests are deployed on the first start.

**Screenshots / Logfiles**
```
root@ledapi4-64:/data/var/containers# kanto-auto-deployer manifests/
Reading manifests from [manifests/]
Wrong json in [manifests/databroker.json]
Wrong json in [manifests/feedercan.json]
Wrong json in [manifests/hvac.json]
Wrong json in [manifests/sua.json]
Wrong json in [manifests/vum.json]
```

**Leda Version (please complete the following information):**
 - Version: 0.0.5
 - Machine:  raspberrypi4-64
 - Connectivity:  transparent internet access

mikehaller, 2023-03-14:
Thank you for reporting this issue. We are aware of a mismatch in the file formats of kanto-auto-deployer vs kantos built-in "init_dir" format.

We'll implement a conversion in kanto-auto-deployer so that it works with both formats.

mikehaller, 2023-03-16:
Issue has been fixed.

closed 2023-03-16 by mikehaller