eclipse-leda/leda-distro #64: RAUC can not install in stream-mode on Raspberry Pi

OlavHerbst, 2023-03-24:
**Describe the bug**
RAUC fails if a bundle from a file server in verity format shall be installed.
The kernel config to enable this is missing even the config file is contained.

**To Reproduce**
Preparation: Flash wic file on a Raspi4 from eclipse-leda-raspberrypi.tar.xz.zip and boot the device

Local bundle:
1. Copy a Rauc bundle in verity-format directly on Raspi. This can be directly one of the bundles in eclipse-leda-raspberrypi.tar.xz.zip, eg. sdv-rauc-bundle-rescue-raspberrypi4-64.raucb
3. Getting info out of bundle succeeds when you call
rauc info sdv-rauc-bundle-rescue-raspberrypi4-64.raucb
4. Installation of local bundle fails:
rauc install sdv-rauc-bundle-rescue-raspberrypi4-64.raucb

Bundle from file server:
1. Connect Raspi to Internet
2. Copy a Rauc bundle in verity-format on a file server. This can be directly one of the bundles in eclipse-leda-raspberrypi.tar.xz.zip, or an already existing one, e.g.: https://olavherbst.de/bin/sdv-sua/sdv-rauc-bundle-rescue-raspberrypi4-64-20230131180730.raucb
3. Getting info out of bundle succeeds when you call
rauc info https://olavherbst.de/bin/sdv-sua/sdv-rauc-bundle-rescue-raspberrypi4-64-20230131180730.raucb
4. Installing the streamed bundle fails:
rauc install https://olavherbst.de/bin/sdv-sua/sdv-rauc-bundle-rescue-raspberrypi4-64-20230131180730.raucb

**Expected behaviour**
Installation of bundles in verity format locally or streamed from file server should succeed.
The kernel should be built with the files, which are already prepared for this stream-feature, in
meta-leda/meta-leda-bsp/dynamic-layers/raspberry-pi/:
  linux-raspberrypi_%.bbappend
  files/leda-bsp-rauc.cfg
The file must be appended to recipe linux-raspberrypi*.bb or be moved to be found and included according to
meta-leda/meta-leda-bsp/conf/layer.conf

**Screenshots / Logfiles**
Installation of local bundle:
`# rauc install sdv-rauc-bundle-rescue-raspberrypi4-64.raucb
...
100% Installing failed.
LastError: Failed mounting bundle: Failed to load dm table: Invalid argument
Installing '/media/sdv-sua/sdv-rauc-bundle-rescue-raspberrypi4-64.raucb' failed
`
Stream-installation from file server:
`# rauc install https://olavherbst.de/bin/sdv-sua/sdv-rauc-bundle-rescue-raspberrypi4-64-20230131180730.raucb
...
100% Installing failed.
LastError: Failed mounting bundle: failed to resolve 'nbd' netlink family - BLK_DEV_NBD not enabled in kernel?
Installing 'https://olavherbst.de/bin/sdv-sua/sdv-rauc-bundle-rescue-raspberrypi4-64-20230131180730.raucb' failed
`

**Leda Version:**
Version: [Commit 62e53c5](https://github.com/eclipse-leda/leda-distro/commit/62e53c5aade96a4e22a4d92e4562893c6cd2ccd2)
    Machine: raspberrypi4-64
    Connectivity: -

mikehaller, 2023-06-21:
`CONFIG_BLK_DEV_NBD` is already set to `y` for Raspi, see [meta-leda/meta-leda-bsp/dynamic-layers/raspberrypi/files/leda-bsp-raspberrypi-rauc.cfg](https://github.com/eclipse-leda/meta-leda/blob/main/meta-leda-bsp/dynamic-layers/raspberrypi/files/leda-bsp-raspberrypi-rauc.cfg#L15)

Is this issue resolved?

OlavHerbst, 2023-06-21:
Yes, this is issue is resolved in this file with, additionally the verity format of the bundle files was set:
CONFIG_DM_VERITY=y

closed 2023-06-29 by mikehaller