eclipse-leda/leda-distro #74: Cannot stop container-maangement service

sophokles73, 2023-04-12:
**Describe the bug**
I have run the latest 0.0.6 image for x86_64 and am not able to restart (nor stop) the container-management service in order to pick up some changes I made in some deployment manifest files.

**To Reproduce**
Steps to reproduce the behaviour:
1. start up leda image in QEMU
2. log in as root
3. wait for systemctl status container-management to indicate that the service is up and running
4. stop the service using `systemctl stop container-management`
5. wait forever ...

**Expected behaviour**
container-management service stops after a few seconds

**Leda Version (please complete the following information):**
 - Version: 0.0.6 as downloaded from GitHub releases
 - Machine: qemux86_64
 - Connectivity: transparent internet access

mikehaller, 2023-04-14:
I have witnessed the same or a similar problem on Raspberry Pi: `systemctl restart container-management` would hang indefinitely. In that case, the device was under heavy load and running on a possibly broken SD-Card (CRC errors when trying to back it up). 

We were using a Kuksa.VAL Seat Service example container reconfigured with physical CAN-Bus access, host networking and in privileged mode. It's unclear what the root cause was, but we could not properly shut down the system any more. The Kanto container-management socket interface was not coming up any more (hence, kantui and kanto-cm were not able to connect to the container-management daemon).

Also stopping the service did not properly work either.
The file `/run/container-management/netns/default` was locked, but unsure whether this is related.

mikehaller, 2026-10-20:
Tested today with a dev build on qemux86_64 with some sdv containers up and running, transparent networking (e.g. ping to external works through qemu slirp -> devcontainer -> wsl -> windows host -> WAN)
Hard to reproduce problem with all the changes in between.

Will close the issue for now, please report if problem appears again.

Log:
```
\u@\h:\w$ kantui
\u@\h:\w$ systemctl restart container-management
\u@\h:\w$ journalctl -xeu container-management
Oct 20 17:40:00 leda-525400123502 systemd[1]: Started Eclipse Kanto - Container Management.
-- Subject: A start job for unit container-management.service has finished successfully
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- A start job for unit container-management.service has finished successfully.
-- 
-- The job identifier is 126.
Oct 20 17:40:00 leda-525400123502 container-management[368]: time="2026-10-20T17:40:00Z" level=info msg="[container-management][daemon_config_util.go:195][pkg:main][func:parseConfigFilePath]  local daemon configuration is set to [/etc/container-management/config.json]"
Oct 20 17:40:10 leda-525400123502 systemd[1]: Stopping Eclipse Kanto - Container Management...
-- Subject: A stop job for unit container-management.service has begun execution
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- A stop job for unit container-management.service has begun execution.
-- 
-- The job identifier is 322.
Oct 20 17:40:23 leda-525400123502 systemd[1]: container-management.service: Deactivated successfully.
-- Subject: Unit succeeded
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- The unit container-management.service has successfully entered the 'dead' state.
Oct 20 17:40:23 leda-525400123502 systemd[1]: Stopped Eclipse Kanto - Container Management.
-- Subject: A stop job for unit container-management.service has finished
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- A stop job for unit container-management.service has finished.
-- 
-- The job identifier is 322 and the job result is done.
Oct 20 17:40:23 leda-525400123502 systemd[1]: container-management.service: Consumed 709ms CPU time over 23.005s wall clock time, 65.5M memory peak.
-- Subject: Resources consumed by unit runtime
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- The unit container-management.service completed and consumed the indicated resources.
Oct 20 17:40:23 leda-525400123502 systemd[1]: Started Eclipse Kanto - Container Management.
-- Subject: A start job for unit container-management.service has finished successfully
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- A start job for unit container-management.service has finished successfully.
-- 
-- The job identifier is 322.
Oct 20 17:40:24 leda-525400123502 container-management[1117]: time="2026-10-20T17:40:24Z" level=info msg="[container-management][daemon_config_util.go:195][pkg:main][func:parseConfigFilePath]  local daemon configuration is set to [/etc/container-management/config.json]"
Oct 20 17:40:24 leda-525400123502 container-management[1117]: subscribed for  edge/thing/response
\u@\h:\w$ 
```

closed 2026-10-20 by mikehaller