eclipse-leda/leda-distro #78: Seat Service Example with CAN-Bus missing candump and incorrect container descriptor

mikehaller, 2023-04-14:
**Describe the bug**
When using the Seat Service example container, it currently has the following bugs:

- Container descriptor misconfiguration: To be able to use the CAN interface, the container must run on the `host` network in use `privileged: true`. The example container descriptor is not usable. Workaround: User has to modify the container descriptor and redeploy.
- Container descriptor misconfiguration: Application environment variables are missing (`SC_CAN=can0` and `CAN=can0`). They should be added to the container.
- Missing candump inside of container: `tools/ecu-reset.sh` is using candump to receive CAN messages. candump is not available inside of the container and the script throws an error message. Workaround: Copy the ecu-reset.sh script from Kuksa.VAL Seat Service Example repository, modify the shebang to use `/bin/sh` instead of `/bin/bash` and run it on the Leda host system.

It may be an idea to have two separate seatservice.json files available (one enabled, the other disabled). So that a user can easily switch between simulated VCAN and physical CAN0.

**Leda Version (please complete the following information):**
 - Version: 0.0.6
 - Machine: raspberrypi4-64

mikehaller, 2023-06-21:
The documentation now contains a chapter how to [make the necessary changes for physical CAN-Bus access](https://eclipse-leda.github.io/leda/docs/app-deployment/seat-adjuster/#hardware-can-bus)

d-s-e, 2023-08-09:
Missing candump was fixed with https://github.com/eclipse/kuksa.val.services/commit/0cb84b8c524e64160b3edf1e607bb4de67e93a11.

closed 2023-10-03 by vasilvas99