eclipse-leda/leda-distro #80: Container with Framebuffer overlay for Leda / SDV / VSS events

mikehaller, 2023-04-14:
**Is your feature request related to a problem? Please describe.**
During hackathons, when there is a display and keyboard attached to the Raspi and the user is on the shell, he needs to switch back and forth between tools (kantui, databroker-cli, logs of containers) to see whether something is working or not.

It would be great to have a way to display something on the screen while the user is on the terminal/shell, like an overlay.
It should not be obstructing too much and still give him access and control over the shell.

**Describe the solution you'd like**
- A tiny application running inside, which listens for Leda / SDV events, such as "Container being deployed" or "Vehicle Signal has changed" or "Application registered datapoints" or "OTA Desired State request received" and displays them in the upper right corner of the screen.
- Maybe another tiny application container, which displays values of VSS signals (numeric, graphical)
- A container with access to the framebuffer device may be sufficient
- A native text user interface interface (like kantui), but which shows Kuksa.VAL Databroker information in "realtime", maybe with a VSS browser and query editor. Like a more advanced version of the "databroker-cli"

**Describe alternatives you've considered**
- A NodeRed dashboard which displays some SDV information, but that requires to run X server + window manager + browser in kiosk mode etc. Sounds too heavyweight.

**Additional notes**
Having a container access to the display or providing a GUI would be an interesting experiment anyway.

d-s-e, 2023-05-03:
A simple lightweight option could be to install `tmux`.
Then it would be possible to have multiple shells/applications running on one screen simultaneously, giving a nice overview while working on the terminal.
You could also add a script, that starts a tmux with a preconfigured set of windows running shells/applications.
Another bonus would be, that this also works over an ssh connection when no display is attached.

vasilvas99, 2023-10-03:
Tmux is now available with https://github.com/eclipse-leda/meta-leda/pull/198

closed 2023-10-03 by vasilvas99