eclipse-leda/leda-distro #99: Kanto cm containers not found in sdv-health

FMT1COB, 2023-05-03:
I have followed this document to run eclipse leda on qemu (https://eclipse-leda.github.io/leda/docs/general-usage/running-qemu/). I have installed qemu in my linux machine and downloaded eclipse-leda-qemu-x86_64.tar.xz and extracted into my working directory. I have created a device id in azure iot hub as mentioned in the document (https://eclipse-leda.github.io/leda/docs/device-provisioning/script-provisioning/). I can find container json files in /data/var/containers/manifests but sdv-health is displaying kanto cm containers are not found.

vasilvas99, 2023-05-03:
All containerized components are pulled during runtime from the internet. Does your qemu image have internet access?

You can also run:

```shell
$ systemctl status kanto-auto-deployer
```

To check the state of the initial container deployment.

FMT1COB, 2023-05-03:
I am getting not found response.
root@qemux86-64:/data/var/containers/manifests# $ systemctl status kanto-auto-deployer
-sh: $: not found
I guess there is not internet. In sdv-health I can see ping internet failed
[SDV Connectivity]
  * Ping [Internet]         : FAILED!	 (error: 1)	 (ping 1.1.1.1)
google.com: resolve call failed: 'google.com' not found
  * DNS Lookup [Internet]   : OK	 ()
  * Cloud Connector         : FAILED!	
  * Device ID:              : FAILED!	 (1)	
------------------------------------------------------

vasilvas99, 2023-05-03:
Hi, it seems you have copied over the `$`, command is only `systemctl status kanto-auto-deployer`. 

Also which version of leda-distro are you using. I suggest the latest milestone release: [0.1.0-M1](https://github.com/eclipse-leda/leda-distro/releases/tag/v0.1.0-M1)

FMT1COB, 2023-05-03:
My Bad!. Yes I am using the same version. Please check  below for the status kanto-auto-deployer
root@qemux86-64:/data/var/containers/manifests# systemctl status kanto-auto-deployer use -l
* kanto-auto-deployer.service - Kanto Auto Deployer
     Loaded: loaded (/lib/systemd/system/kanto-auto-deployer.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2023-05-03 12:54:50 UTC; 13min ago
   Main PID: 543 (kanto-auto-depl)
      Tasks: 10 (limit: 2371)
     Memory: 1.4M
     CGroup: /system.slice/kanto-auto-deployer.service
             `- 543 /usr/bin/kanto-auto-deployer /data/var/containers/manifests --daemon

May 03 12:55:02 qemux86-64 kanto-auto-deployer[543]: [2023-05-03T12:55:02Z WARN  kanto_auto_deployer::manifest_parser] Failed to load manifest directly. Will attempt auto-conversion from init-dir format.
May 03 12:55:02 qemux86-64 kanto-auto-deployer[543]: [2023-05-03T12:55:02Z INFO  kanto_auto_deployer] Creating [seatservice-example]
May 03 12:55:03 qemux86-64 kanto-auto-deployer[543]: [2023-05-03T12:55:03Z ERROR kanto_auto_deployer] [CM error] Failed to create container: status: Unknown, message: "failed to resolve reference \"ghcr.io/boschglobal/kuksa.val.services/seat_service:v0.3.0\": failed to do request: Head \"https://ghcr.io/v2/boschglobal/kuksa.val.services/seat_service/manifests/v0.3.0\": dial tcp: lookup ghcr.io on 10.0.2.3:53: no such host", details: [], metadata: MetadataMap { headers: {"content-type": "application/grpc"} }
May 03 12:55:03 qemux86-64 kanto-auto-deployer[543]: [2023-05-03T12:55:03Z WARN  kanto_auto_deployer::manifest_parser] Failed to load manifest directly. Will attempt auto-conversion from init-dir format.
May 03 12:55:03 qemux86-64 kanto-auto-deployer[543]: [2023-05-03T12:55:03Z INFO  kanto_auto_deployer] Creating [sua]
May 03 12:55:04 qemux86-64 kanto-auto-deployer[543]: [2023-05-03T12:55:04Z ERROR kanto_auto_deployer] [CM error] Failed to create container: status: Unknown, message: "failed to resolve reference \"ghcr.io/eclipse-leda/leda-contrib-self-update-agent/self-update-agent:build-66\": failed to do request: Head \"https://ghcr.io/v2/eclipse-leda/leda-contrib-self-update-agent/self-update-agent/manifests/build-66\": dial tcp: lookup ghcr.io on 10.0.2.3:53: no such host", details: [], metadata: MetadataMap { headers: {"content-type": "application/grpc"} }
May 03 12:55:04 qemux86-64 kanto-auto-deployer[543]: [2023-05-03T12:55:04Z WARN  kanto_auto_deployer::manifest_parser] Failed to load manifest directly. Will attempt auto-conversion from init-dir format.
May 03 12:55:04 qemux86-64 kanto-auto-deployer[543]: [2023-05-03T12:55:04Z INFO  kanto_auto_deployer] Creating [vum]
May 03 12:55:05 qemux86-64 kanto-auto-deployer[543]: [2023-05-03T12:55:05Z ERROR kanto_auto_deployer] [CM error] Failed to create container: status: Unknown, message: "failed to resolve reference \"ghcr.io/eclipse-leda/leda-contrib-vehicle-update-manager/vehicleupdatemanager:main-1d8dca55a755c4b3c7bc06eabfa06ad49e068a48\": failed to do request: Head \"https://ghcr.io/v2/eclipse-leda/leda-contrib-vehicle-update-manager/vehicleupdatemanager/manifests/main-1d8dca55a755c4b3c7bc06eabfa06ad49e068a48\": dial tcp: lookup ghcr.io on 10.0.2.3:53: no such host", details: [], metadata: MetadataMap { headers: {"content-type": "application/grpc"} }
May 03 12:55:05 qemux86-64 kanto-auto-deployer[543]: [2023-05-03T12:55:05Z INFO  kanto_auto_deployer] Running in daemon mode. Continuously monitoring "/data/var/containers/manifests"
Unit use.service could not be found.

vasilvas99, 2023-05-03:
Are you perhaps behind a corporate proxy, since the run-leda.sh script sets-up slirp-networking and requires transparent internet access?

FMT1COB, 2023-05-04:
I have updated the proxy using export command

vasilvas99, 2023-05-04:
It seems that the proxy is denying you access to a public GitHub registry.

Can you try two things:
1. `wget google.com` - to check if DNS resolution/internet access works in general
2. Do you have docker on your host machine? If yes, can you (using the same proxy settings) execute: 
```
docker pull ghcr.io/eclipse-leda/leda-contrib-cloud-connector/cloudconnector:main-47c01227a620a3dbd85b66e177205c06c0f7a52e
```

closed 2023-05-22 by mikehaller