eclipse-leda/leda-distro #105: No WiFi on raspberry4

sophokles73, 2023-05-22:
**Describe the bug**
Leda image 0.1.0-M2 on a Raspberry4B does not connect to WiFi

**To Reproduce**
1. follow the steps to write the image to an SD card as described in https://eclipse-leda.github.io/leda/docs/general-usage/raspberry-pi/
2. Add a `wpa_supplicant.conf` file to the boot partition with the following content
   ```
   ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
   update_config=1
   country=DE

   network={
        scan_ssid=1
        ssid="my_ssid"
        psk="my_wpa2_pwd"
   }
   ```
3. Boot Raspi from SD card
4. Verify that Raspi has not connected to WiFi using router Web UI

**Expected behaviour**
Raspi should have connected to WiFi and should show up in router's Web UI

**Screenshots / Logfiles**

**Leda Version (please complete the following information):**
 - Version: 0.1.0-M2
 - Machine: raspberrypi4-64
 - Connectivity: local WiFi with transparent internet access

stlachev, 2023-05-23:
Hi Kai,
According to our documentation [Headless Wi-Fi configuration](https://eclipse-leda.github.io/leda/docs/general-usage/wifi-configuration/connecting-to-wifi/), you can use:
```
ctrl_interface=/var/run/wpa_supplicant
ctrl_interface_group=0
update_config=1
network={
        key_mgmt=NONE
}
network={
    ssid="my_ssid"
    psk="my_wpa2_pwd"
    key_mgmt=WPA-PSK
}
```
The file can be created by Notepad++ as a Unix one (LF only). 
If you already saw that your previous file (wpa_supplicant.conf) does not work, you can delete it from /etc and /boot folder. 
You can also create the file as described in step [Connecting to a network](https://eclipse-leda.github.io/leda/docs/general-usage/wifi-configuration/connecting-to-wifi/) just to see if it is working for you, and then move it to /boot partition or distribute it if you like to, or just keep it there in /etc. Note that the line:
`wpa_passphrase <SSID> <Passphrase> >> /etc/wpa_supplicant.conf` appends

sophokles73, 2023-05-23:
Hey @stlachev thanks for the quick response. I failed to see the dedicated section on WiFi configuration in the docs (stupid me ;-)) In any case, I had been following the guide at https://eclipse-leda.github.io/leda/docs/general-usage/raspberry-pi/ and there is one line

> Optional: If you need to adapt the network configuration eg Wifi credentials, edit the configuration files on the boot partition.

which IMHO could benefit from including a link to https://eclipse-leda.github.io/leda/docs/general-usage/wifi-configuration/ ...

closed 2023-05-23 by sophokles73

sophokles73, 2023-05-24:
I have update my config according to the Leda documentation and everything works as expected now. Thanks again for your help @stlachev :+1: