eclipse-leda/leda-distro #111: Additional debug tools in default image, e.g. strace

mikehaller, 2023-06-14:
During onboarding of Leda distro to new hardware devices, such as the Jacinto eval kit, it's necessary to go through a couple of low-level functionality: boot process, filesystem layouts, RAUC integration etc.

Often, it's not clear when some tools have problems, e.g. "fw_printenv" not finding the correct config file and problems like that.

Tools like `strace` would be helpful to have on the default full image (sdv-image-full), to be able to debug the system. Maybe a "-dev" feature toggle. The alternative option to install these tools as a container is not practical, as the state of the system integration is very early and at that point in time, the container runtime may not even be functional yet.

d-s-e, 2023-08-09:
The full image is used for hacking anyway, so I don't think the feature toggle is necessary.
Installing just `strace`is easy, which other tools do you need?

Usually `EXTRA_IMAGE_FEATURES:append = " tools-debug"` is used for that, but this will fail due to the GPLv3 restriction.

vasilvas99, 2023-10-03:
Strace was added in https://github.com/eclipse-leda/meta-leda/pull/202 , its small enough (~2MBs). Maybe we can also add file and ldd and this should be more than enough for basic on-device debugging of binaries?

closed 2023-11-27 by mikehaller