eclipse-leda/leda-distro #136: Leda on Red Hat In-Vehicle OS (RHIVOS)

mikehaller, 2023-07-27:
The Leda distro (the EDGE stack components) should be supporting Red Hat as a base OS, as many OEMs plan to use RHIVOS for their Vehicle Computer In-Vehicle Linux domain.

To be checked:
- How to package (Snap, Flatpak, rpm?)
- Which container runtime comes ootb and how to make Kanto work on top of it

mikehaller, 2023-07-27:
References:
- https://sig.centos.org/automotive/#automotive-stream-distribution
- https://autosd.sig.centos.org/
- https://sig.centos.org/automotive/content_definition/
- https://wiki.centos.org/SpecialInterestGroup/Automotive/Meetings

ericcurtin, 2024-02-12:
> How to package (Snap, Flatpak, rpm?)

We generally recommend rpms or podman containers for software packaging.

> Which container runtime comes ootb and how to make Kanto work on top of it

crun and podman.