eclipse-leda/leda-distro #146: [ISSUE] The latest cannot work on Leda v0.1.0-M1 based on QEMU X86_64.

ley901017, 2023-09-12:
Hi，

I build a container based on [Vehicle App cpp Template](https://github.com/eclipse-velocitas/vehicle-app-cpp-template) in Codespace.
I pushed the [container](https://hub.docker.com/r/lenyanc/vehicle_cpp_app/tags) named `lenyanc/vehicle_cpp_app:0.2.0` into docker hub, then I deploy it on my Qemu with `kanto-auto-deployer`. The below issue has been raised:
```
terminate called after throwing an instance of 'mqtt::exception'
  what(): MQTT error [-14]: Invalid protocol scheme
```
The JSON file content is as below.
```json
{
    "id": "",
    "name": "vehicle-cpp-app",
    "image": {
        "name": "docker.io/lenyanc/vehicle_cpp_app:0.2.0",
        "decrypt_config": null
    },
    "host_name": "",
    "domain_name": "",
    "resolv_conf_path": "",
    "hosts_path": "",
    "hostname_path": "",
    "mounts": [],
    "hooks": [],
    "host_config": {
        "devices": [],
        "network_mode": "host",
        "privileged": true,
        "restart_policy": {
            "maximum_retry_count": 0,
            "retry_timeout": 0,
            "type": "unless-stopped"
        },
        "runtime": "io.containerd.runc.v2",
        "extra_hosts": [
        ],
        "port_mappings": [
        ],
        "log_config": {
            "driver_config": {
                "type": "json-file",
                "max_files": 2,
                "max_size": "1M",
                "root_dir": ""
            },
            "mode_config": {
                "mode": "blocking",
                "max_buffer_size": ""
            }
        },
        "resources": null
    },
    "io_config": {
        "attach_stderr": false,
        "attach_stdin": false,
        "attach_stdout": false,
        "open_stdin": false,
        "stdin_once": false,
        "tty": false
    },
    "config": {
        "env": [
	   "SDV_MIDDLEWARE_TYPE=native",
	   "SDV_MQTT_ADDRESS=mqtt://172.17.0.1:1883",
	   "SDV_VEHICLEDATABROKER_ADDRESS=grpc://172.17.0.1:30555",
           "SDV_SEATSERVICE_ADDRESS=grpc://172.17.0.1:30051",
           "RUST_LOG=debug",
           "vehicle_data_broker=debug"
        ],
        "cmd": []
    },
    "network_settings": null,
    "state": {
        "pid": -1,
        "started_at": "",
        "error": "",
        "exit_code": 0,
        "finished_at": "",
        "exited": false,
        "dead": false,
        "restarting": false,
        "paused": false,
        "running": false,
        "status": "",
        "oom_killed": false
    },
    "created": "",
    "manually_stopped": false,
    "restart_count": 0
}
```

I am not sure if it is a bug, please help me give some solutions or suggestions.

vasilvas99, 2023-09-12:
Hi, could you remove the protocol specifiers from your env variables, e.g.:

```
        "env": [
	   "SDV_MIDDLEWARE_TYPE=native",
	   "SDV_MQTT_ADDRESS=172.17.0.1:1883",
	   "SDV_VEHICLEDATABROKER_ADDRESS=172.17.0.1:30555",
           "SDV_SEATSERVICE_ADDRESS=172.17.0.1:30051",
           "RUST_LOG=debug",
           "vehicle_data_broker=debug"
        ],

```

This was enough to get the container running on my test setup (haven't explored much further than that, so there might be more issues)

ley901017, 2023-09-15:
Thanks  for your support. Your solution can work very well.

closed 2023-09-15 by vasilvas99