eclipse-leda/leda-distro #147: Support for WasmEdge WebAssembly

mikehaller, 2023-09-12:
From the wasmedge website: WasmEdge is a lightweight, high-performance, and extensible WebAssembly runtime for cloud-native, edge, and decentralized applications. Main use case for WasmEdge is to safely execute (thirdparty) code as plug-ins in a software product.

This sounds like a matching use case for an SDV Edge runtime platform like Eclipse Leda.

With WasmEdge, Leda would be able to execute workload not only as Docker/OCI containers, but could also support wasm32 packages.

By combining WasmEdge with crun as the low-level container runtime for containerd, Leda would also gain the benefit of lower memory consumption, higher performance and the rest of the stack could remain the same (e.g. Kanto as container orchestrator).

References:
- POC: https://github.com/SoftwareDefinedVehicle/leda-distro-fork/tree/wasmedge
- https://github.com/WasmEdge/WasmEdge
- https://wasmedge.org
- https://wasmedge.org/docs/develop/deploy/cri-runtime/containerd-crun

vasilvas99, 2023-09-13:
Related: https://github.com/eclipse-leda/meta-leda/pull/192

vasilvas99, 2023-09-18:
Compiling https://github.com/RustPython/RustPython for the wasm32-wasmi target on a laptop and taking it to the arm64 image, where it's is compiled to a native binary with wasmedge compile --optimize 3 --enable-all rustpython.wasm rustpython leads to a functioning RustPython binary with the standard library included

vasilvas99, 2023-11-03:
Partially resolved with: https://github.com/eclipse-leda/meta-leda/pull/192 as far as Kanto CM allows

closed 2023-11-03 by vasilvas99