eclipse-leda/leda-distro #151: eclipse-leda-raspberrypi.tar.xz file containing image with boot partition is FAT16 and not FAT32

christian-kahr-itk, 2023-09-14:
**Describe the bug**

We downloaded https://github.com/eclipse-leda/leda-distro/releases/download/v0.1.0-M2/eclipse-leda-raspberrypi.tar.xz
and tried to flash it to SD card according to https://eclipse-leda.github.io/leda/docs/general-usage/raspberry-pi/ 
but we always got cannot read partition 0 as FAT. 

So we checked the partition type and saw it's not FAT32. Linux say it's FAT16 - Windows 7-ZIP identifies it as Windows BDP. 

**To Reproduce**
Steps to reproduce the behaviour:
1. Download *.tar.xz
2. flash tar.xz or wic 

**Expected behaviour**
a image to use on RPI 4

**Screenshots / Logfiles**
[image]

**Leda Version (please complete the following information):**
 - Version: v0.1.0-M2
 - Machine: raspberrypi4-64

vasilvas99, 2023-09-18:
Hi, I am failing to reproduce the bug.  7zip shows a similar picture as the one you've posted:
[image].

My workflow is as follows (Ubuntu 22.04):
1. Download the tar (*note the tar.xz cannot be flashed directly*) with:
```
wget https://github.com/eclipse-leda/leda-distro/releases/download/v0.1.0-M2/eclipse-leda-raspberrypi.tar.xz
```
2. Extract the archive with: 
```
 tar -xvf eclipse-leda-raspberrypi.tar.xz -C leda && cd leda
```
3. Install bmaptools (ref: https://command-not-found.com/bmaptool)
4. Flash with bmaptool :
```
sudo bmaptool copy --bmap sdv-image-all-raspberrypi4-64.wic.bmap sdv-image-all-raspberrypi4-64.wic.bz2 /dev/mmcblk0
```

Since you mentioned it, I assume you have a Linux machine available, could you try exactly the steps above? Only difference might be that you might have to substitute `/dev/mmcblk0` with the path to the device.

Could you try with a different sd card?

christian-kahr-itk, 2023-09-18:
@vasilvas99 will give it a try when find again time :D but why didnt you mention in documentation bmaptool?

vasilvas99, 2023-09-18:
> @vasilvas99 will give it a try when find again time :D but why didnt you mention in documentation bmaptool?

Thanks :)
Bmaptool is mentioned in the doc page you've linked a bit further down:
[image]

christian-kahr-itk, 2023-09-18:
ok i will give it a try - i asked my collegue and after seeing it pretty sure we tried too. just one small question. is the tests done on PI3 or PI4? because PI3 can handle FAT16 boot partition but PI4 cannot handle it.

vasilvas99, 2023-09-18:
RPi4B-64 is the only one currently supported. My test board is one of those as well.

Thanks for the the help with debugging

mikehaller, 2026-10-20:
I found a [release note in the RPI bootloader documentation](https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware-2711/release-notes.md#2021-10-04---add-support-for-gpt-fat16-and-increase-usb-timeouts---beta) which might explain the FAT16/FAT32 confusion: FAT16 support was added after FAT32 support, in 2021, so maybe the firmware was older than 2021 and thus didn't support FAT16 - and the ones we were using for testing were newer and thus supported FAT16.

Anyway, we'll try to force the VFAT partitions to be FAT32 now, which may have other downsides (e.g. minimum cluster size or partition size). Unfortunately, i won't be able to build and test the raspi images on GitHub Actions, as the required self-hosted infrastructure for the huge BitBake build has been shut down / reallocated to other projects with much higher priority.

closed 2026-10-20 by mikehaller